julianschill/klipper-led_effect #159: Klipper showing dirty

SHoover80, 2023-10-09:
I have two printers and noticed since I started using led_effect it's showing: Host(aarch64, 64bit)
Version: v0.11.0-295-g83ef0e13-dirty. Thinking it was an issue with Klipper I reflashed my host and it's normal. But once I install led_effect again I see that klipper is showing dirty again.  Is this normal? something to be concerned about? am I doing something wrong?

julianschill, 2023-10-09:
To work, led_effect has to put a file (more exactly a link to a file) in the klipper directory. This leads to the repository being "dirty" and is normal.

SHoover80, 2023-10-09:
Ahh, ok. I figured it would be something like that. I just wanted to be sure. I appreciate the response. @julianschill

closed 2023-10-11 by SHoover80