julianschill/klipper-led_effect #161: Repo has untracked source files: ['klippy/extras/led_effect.py']

JNSmith1717, 2023-10-20:
like it says under Update Manager under klipper it has "Repo has untracked source files: ['klippy/extras/led_effect.py']"

JNSmith1717, 2023-10-20:
[image]

ayesim, 2023-10-20:
it is normal. klipper has additional files in klipper installation folder and warning you. it's not an error message.

JNSmith1717, 2023-10-20:
Yeah i know just annoying, is there a way to get rid of this message?  Also is there a way to uninstall the led_effect?  If this continues to show this I would rather just uninstall the led_effect.

julianschill, 2023-10-20:
There is no way to get rid of that message. It's just how the plugin extends Klipper. Same method is used for all Klipper "plugins". To get rid of the message you have to uninstall it. Just delete the file `~/klipper/klippy/extras/led_effect.py`.

JNSmith1717, 2023-10-23:
> There is no way to get rid of that message. It's just how the plugin extends Klipper. Same method is used for all Klipper "plugins". To get rid of the message you have to uninstall it. Just delete the file `~/klipper/klippy/extras/led_effect.py`.

Thank you, I think I will just delete, nothing agaisnt what you did, I think its a great Idea but honestly I don't even use it.  I wish I could have found none RGB lights for the stealth burner.

JNSmith1717, 2023-10-26:
Well i deleted ~/klipper/klippy/extras/led_effect.py and it still says the Repo has untrack source files for this, is there a fix for this?

julianschill, 2023-10-26:
Try refreshing the repos (there is a button in Mainsail) or rebooting and reloading the page. Also make sure it is led_effect.py it reports as untracked and not some other plugin.