julianschill/klipper-led_effect #162: KlipperScreen LED Panel issue

Giadej, 2023-10-20:
I've installed  klipper-led effect to control the neopixels on my printer and everything (effects and templates) work perfectly. 

But if I want to change the color of the lights from the klipperscreen LEDS panel I can't do it. 
The color change occurs for half a second and then returns to the previous effect/template used. 

If I stop the effects (from the console command with STOP_LED_EFFECTS), I can use the klipperscreen's panel to change the color of the lights from the display! 

There is a way to adjust the color of the lights without having to send  the STOP_LED_EFFECTS command?

julianschill, 2023-10-20:
When an Effect is active the LEDs are controlled by the plugin until you stop that effect. So there is unfortunately no way to do both at the same time.

Giadej, 2023-10-21:
> When an Effect is active the LEDs are controlled by the plugin until you stop that effect. So there is unfortunately no way to do both at the same time.

I understand now. 
Thank you for the reply and for your amazing plugin.

closed 2023-10-21 by Giadej