julianschill/klipper-led_effect #163: Combination of heater and progress

testack18, 2023-10-26:
Hi,

I would like to create an effect where all the leds are blue and transits to red as the chamber temp goes up. Additionally would like not to transit the whole strip but increase the number of leds which color are changed. Let me say I have 10 LEDs on the strip. My macro waits for the temp sensor of chamber to goes up to 50C. For simplicity of the description lets say the current temp is 30C. That would mean every 2C increasement an additional LED would turn to red while the rest remaining blue.
I have tried the heater effect but it turns the whole strip's color. 
What I want is something like a mixture of progress and heater. The effect goes like in case of the Progreess but is bonded to a heater or temp sensor.

Is that possible?

Thank you in advance!

Kind regards,
Gabor

julianschill, 2023-10-26:
It's not possible yet, but such effects are on the todo list. It is tracked by issue #25

testack18, 2023-10-26:
Thanks! The linked issue is really old. Do you have any rough estimation when you will have time to implement it?

julianschill, 2023-10-26:
No. I do this in my free time, whenever I feel like doing it. There is absolutely no time planning.

julianschill, 2024-04-28:
Closing this as it is duplicate to #25

closed 2024-04-28 by julianschill