julianschill/klipper-led_effect #164: LED freezing

appleimperio, 2023-11-09:
If I run an effect everything run smoothly as soon as I move the steppers motors the effect start freezing until the motor stop. I read your troubleshooting and see the steppers can cause interference so to make the test I left the cables as is and use the voltage of the main board but with an Esp8266 board install WLED and test. With the cable on the same route connected to the Esp8266 I didn't see the freeze when the motors move. Since I have two lines of LED I left one on the main board and one on the Esp8266 with WLED the on the board using Klipper_led_effect replicate the problem the one with WLED run smoothly. Is there any other troubleshooting? or maybe my configurations is wrong.

```
[neopixel disco1]
pin: PA9
chain_count: 25
color_order: GRB
initial_RED: 1.0
initial_GREEN: 1.0
initial_BLUE: 1.0

[neopixel disco2]
pin: PB15
chain_count: 25
color_order: GRB
initial_RED: 1.0
initial_GREEN: 1.0
initial_BLUE: 1.0

```

SYNCRAFT-GITHUB, 2023-11-23:
Hi, if you are using a manta m8p board that might be the problem. I can't tell for sure, but it seems to be related to processor timing. I had no issues with the STM32F407 micro-controller, but have experienced this with the STM32GB01.

appleimperio, 2023-11-24:
That’s the one I’m using. I ended up using WLED. Thank you!

closed 2023-11-24 by appleimperio