julianschill/klipper-led_effect #171: uninstall

bybarbaros, 2024-01-28:
How can I uninstall

julianschill, 2024-02-10:
Delete the file ~/klipper/klippy/extras/led_effect.py and remove all your config entries. Additionally delete the folder ~/klipper-led_effects

bybarbaros, 2024-02-11:
Thank you Julian... I deleted the files. and reinstalled. I was able to run
the macro you made. I use a single LED bar in the system. There are 72
cells. I did this by sharing the LED definition among the relevant macros.
I can change the system again when I make a stealburner hotend in the
future. I encountered 2 problems. A problem occurs like in the video I
uploaded to YouTube. When I open the application, there is a white flash
for a moment and the same thing happens with other macros and colors. In
the second case, I added a process macro in the start print macro I added
to the system before the nozzle cleaning process and added the command to
stop the macro when it finished. However, the print start macro does not
complete the nozzle cleaning during operation (halfway through the process).
  LED macro is changing. I couldn't understand the reason for this. I'm
adding the relevant macros below. I would be very grateful if you could
help.

https://www.youtube.com/watch?v=5Ju65cpjN5U

[neopixel sb_leds]
pin: PC5
#   The pin connected to the neopixel. This parameter must be provided.
chain_count: 72
#   The number of Neopixel chips that are "daisy chained" to the
#   provided pin. The default is 1 (which indicates only a single
#   Neopixel is connected to the pin).
color_order: GRB   #   GRB, GRB, GRB, GRB, GRB, GRB, GRB, GRBW,  GRBW
#   Set the pixel order required by the LED hardware. Options are GRB,
#   RGB, GRBW, or RGBW. The default is GRB.
initial_RED: 1.0
initial_GREEN: 0.0
initial_BLUE: 1.0
#initial_WHITE: 0.0
#   Sets the initial LED color of the Neopixel. Each value should be
#   between 0.0 and 1.0. The WHITE option is only available on RGBW
#   LEDs. The default for each color is 0.#

##########################
# LED Effects Animations #
##########################

##################
## logo effects ##
##################

[led_effect sb_logo_busy]
autostart:              false
frame_rate:             24
leds:
    neopixel:sb_leds (25-50)
layers:
    breathing  3 1 top (1,0,0)

[led_effect sb_logo_cleaning]
autostart:              false
frame_rate:             24
leds:
    neopixel:sb_leds (25-50)
layers:
    breathing  3 1 top (0.0, 0.02, 0.5)

[led_effect sb_logo_calibrating_z]
autostart:              false
frame_rate:             24
leds:
    neopixel:sb_leds (25-50)
layers:
        breathing  3 1 top (0.0, 0.0, 0.35)

[led_effect sb_logo_heating]
leds:
    neopixel:sb_leds (25-50) # (1,3,7,8,6,2,63,65,71,72,70,64)
[... 322 lines cut ...]
 SET_LED_EFFECT EFFECT=sb_logo_cleaning

 G90                                            ; absolute positioning
 ## Move nozzle to start position
 G1 X{start_x} Y{start_y} F6000
 G1 Z{start_z} F1500

 ## Wipe nozzle
 {% for wipes in range(1, (wipe_qty + 1)) %}
   G1 Y{start_y + wipe_dist} F{wipe_spd * 60}
   G1 Y{start_y} F{wipe_spd * 60}
 {% endfor %}

 ## Raise nozzle
 G1  Z10 F1000 #Z{raise_distance}

  STOP_LED_EFFECTS
  SET_LED_EFFECT EFFECT=rainbow

SAYGILARIMLA

BARBAROS DOĞANCI

Julian Schill ***@***.***>, 10 Şub 2024 Cmt, 10:13 tarihinde
şunu yazdı:

burntcoinpurse, 2024-02-11:
Completely un-readable on github lol. But I can read it in the email notification. Try commenting out the case light to see if that helps. You can also try condensing the leds i.e. 

`[led_effect sb_logo_heating] 
leds: 
neopixel:sb_leds (25-50) # (1,3,7,8,6,2,63,65,71,72,70,64) 
autostart: false 
frame_rate: 24 
layers: 
chase 0.30 0.60 add (1.00,1.00,1.00),(0.0,0.0,0.0) #comet 0.3 0 
add (1, 0.18, 0)`

This could also be causing an issue

julianschill, 2024-02-21:
The white flashes I have seen a few times and I think it's from an overload of the MCU. Unfortunately the neopixel control implementation in klipper is not very efficient and can get interrupted by e.g. motor stepping. This might cause these flashes. It helps to put all leds on a separate MCU (e.g. a Raspberry Pi Pico).
For the nozzle cleaning I am not sure what you mean by not completing. The effects shouldn't have any influence on the movement of the printer.

bybarbaros, 2024-02-24:
Let me try to explain. for example. Let's say when the print start macro starts. Let's assign an LED effect. hotend home did it all. qgl did it. He turned on the heaters. From here on it will start cleaning the nozzle. (In the meantime, I set a different led effect macro.) and while the nozzle cleaning started, the led effect macro changed. But before the cleaning macro is finished, the led effect returns to the first led effect macro I set. Can timing setting be added for led effect here? To make the runtime of the clean nozzle macro and the led effect macro the same. note: please excuse any spelling errors and misunderstandings. I use google taranslate. I often lose meaning in translating what I want to say and your answers.

closed 2024-04-28 by julianschill