julianschill/klipper-led_effect #172: STATUS_CALIBRATING_Z

1kcender, 2024-02-10:
During STATUS_CALIBRATING_Z  Nozzle lights 1st come on bright white and after picking up klicky switch they go to a different status while probing endstop, Switch and bed. Then come back on as it goes to dock probe. I am thinking this has something to do with how the config is written for calibrate_Z. Thanks.

julianschill, 2024-02-10:
Not sure why that is. Shouldn't behave that way. Can you share your log please?

1kcender, 2024-02-11:
Sure Thing. 
[klippy.log](https://github.com/julianschill/klipper-led_effect/files/14231026/klippy.log)
Thia what you needed?

julianschill, 2024-02-21:
Not sure what is going on there. I suspect a hardware problem somehow. The effect shouldn't react on the klicky at all. Just logo breathing and nozzle leds on.

1kcender, 2024-02-21:
Thank you for taking a look at it. I have Taken a video to better explain what I am seeing. During Home and Z-tilt it does have the Nozzle lights on, which i believe to be the intended operation. During bed mesh and Calibrate Z, the lights go to a breathing rainbow during probing. 
Again Thanks for having taken the time to look at this.

[https://www.youtube.com/watch?v=Kcw23BDAidc](url)

julianschill, 2024-02-22:
You have Macros like [gcode_macro _klicky_status_ready] which are called while docking and attaching the klicky and call the SB Status macros. Those are causing the effect changes. So not a problem of this plugin but by your macros.

1kcender, 2024-02-24:
Thank you. I will close issue and look into fixing this thru klicky macros.

closed 2024-02-24 by 1kcender

julianschill, 2024-02-24:
Thanks. Much appreciated.