julianschill/klipper-led_effect #186: Simulator will not simulate white LED on RGBW / GRBW chains

ajaytanna, 2024-04-04:
As the title says, adding the fourth element of white does not simulate the effect in the latest simulator.

closed 2024-04-05 by julianschill

ajaytanna, 2024-04-06:
Ah! My monitor only shows RGB!! That's why I have been unable to see any whites and pastels or pink and peach. Not to mention turquoise and emerald!! You are so very right! I just put a 40X magnifier to the screen. Indeed its is RGB only. Just the way it was 50 years back when i tried it as a kid.
Thanks for putting things in the correct chromatic perspective.

julianschill, 2024-04-06:
Not sure if you are being sarcastic or not. Your monitor shows only RGB and mixes all colors with them. White ist just all colors mixed together. You can do the same with LEDs. Some LEDs have an additional white LED which shows a better spectrum of white. The color temperature of this LED varies and you can choose between warm, neutral and cold white.
Monitors don't have a separate white channel I could set the 4th value to. I could mix it into the other colors, but that would also not reflect reality. I could show a white ring around the circles, which also wouldn't look great. So I chose to just ignore the white channel in the simulator, which would reflect the behavior of RGB only LEDs like the WS2812.

ajaytanna, 2024-04-07:
Oh Mon dieu!
It's colour model and colour spaces you are talking about not the monitor per se. Just like Hexachrome once in printing used to be CMYK+O+G (Orange and Green) elements added together for better fidelity in printing. Today Epson and several other vendors mix light cyan and light magenta to CMYK in photo printers. Just so White is added to RGB to create several other shades beyond the gamut of RGB. How that is represented on screen is for people like you (scientists) to imagine and invent. I respect that you do not want to bother yourself with it at the moment. But I sincerely appreciate the work you have done and shared.
Cheers!