julianschill/klipper-led_effect #188: Delta printer - Homing effect

maxduracell, 2024-04-20:
Its possible that I have something defined incorrectly, but I have a delta that I'd like to trigger a flash while homing.  It will only flash for the final tower, whichever that is based on the position of the toolhead.  Is it possible to get it to flash for all 3?  Here is a bit of my config.  Assume A, B, and C steppers are more or less identically setup with their appropriate pins.  The probe flashes without issue.

[stepper_c] # Z
step_pin: PB5
dir_pin: PB4
enable_pin: !PB8
microsteps: 128
rotation_distance: 40
full_steps_per_rotation: 200
endstop_pin: ^PC4 
homing_speed: 60
homing_retract_dist: 5.0
homing_retract_speed: 10 

[led_effect homing]
leds:
    neopixel:lights
endstops:  a, b, c, probe
autostart: true
frame_rate: 24
layers:
    homing 5 0 top (1,1,1)

julianschill, 2024-04-27:
I look what I can do, but unfortunately I don't have a delta printer.

julianschill, 2024-04-28:
@maxduracell  Please run a homing sequence with the effect enabled and post the klippy.log here.

maxduracell, 2024-04-28:
I think I captured it.  
[klippy (11).log](https://github.com/julianschill/klipper-led_effect/files/15142754/klippy.11.log)

julianschill, 2024-04-28:
Try creating three separate effects and assign each endstop to one of the effects and let me know if that would solve it. I think the issue is the simultaneous homing of the axes.

julianschill, 2024-04-28:
Just checked the Klipper code and it seems that the "homing:homing_move_end" event is only triggered after all axes are homed. So I fear I can't do much here without patching Klipper.

maxduracell, 2024-04-28:
> Try creating three separate effects and assign each endstop to one of the effects and let me know if that would solve it. I think the issue is the simultaneous homing of the axes.

I tried this.  No good.  It was interesting, though, that it combined the colors of all 3 effects.  It is what it is.  Thanks for having a look at it.

julianschill, 2024-04-28:
I found a way to react to buttons. So you could use a duplicate pin to react on that pin. If you know how to use a branch you can try it out here: https://github.com/julianschill/klipper-led_effect/tree/feature/buttons

I will merge that after I did some more testing and have written the documentation.

It works something like that: 

`[duplicate_pin_override]
pins: PC1,PC2

[led_effect button]
autostart: true
button_pins: PC1,PC2
leds:
  neopixel:chamber_light
layers:
  flashbutton 0 0.5 top (1,1,1)`
  
Let me know if this works for you.

maxduracell, 2024-04-29:
I've haven't yet had to use a branch, but I'll see if I can work it out.  Might be a couple days before I get back around to it.

maxduracell, 2024-05-01:
I evidently don't know how to install branches.  I keep getting repository not found even though I can clearly see it.

maxduracell, 2024-05-01:
Its working perfectly!  Thanks for taking the time on this.

closed 2024-05-18 by julianschill