julianschill/klipper-led_effect #199: Error gcode_macro suite a mise a jour

mickael13160, 2024-05-20:
unknown config object 'gcode_macro' once the underlying issue is corrected, use the "restart" command to reload the config and restart the host software. printer is halted site:www.reddit.com

julianschill, 2024-05-20:
Ok, I think I see the problem. Your led_effects are at the very beginning of the config, so that noch gcode_macro has been loaded yet. I have to check what I can do there. As a workaround move the led_effects (I think you have them as an include) further down in the config after a gcode_macro definition. Or just put this at the very beginning of your printer.cfg:
```
[gcode_macro _workaround]
gcode: 
```

This will load the gcode_macro module, that is needed by led_effect.

julianschill, 2024-05-20:
Should be fixed now. Please update klipper-led_effect

mickael13160, 2024-05-20:
thanks now I have another error message
Section 'mmu_led_effect mmu_breathing_red' is not a valid configuration section
[klippy.log](https://github.com/julianschill/klipper-led_effect/files/15378030/klippy.log)

julianschill, 2024-05-20:
This seems to come from HappyHare. Make sure you have that installed correctly and contact them for problems.

mickael13160, 2024-05-20:
Thank you very much for your help I will look into this and contact them

mickael13160, 2024-05-20:
good evening problem solved I forgot to update happy hare via the command line after having updated via mainsail

julianschill, 2024-05-20:
Great! I will close the issue then.

closed 2024-05-20 by julianschill