julianschill/klipper-led_effect #202: Simulator won't work

imkebe, 2024-05-26:
MacOS venv

```
❯ ./bin/python --version
Python 3.10.14

❯ ./bin/pip list
Package       Version
------------- --------
coloredlogs   15.0.1
flatbuffers   24.3.25
humanfriendly 10.0
mpmath        1.3.0
numpy         1.26.4
onnxruntime   1.18.0
opencv-python 4.9.0.80
packaging     24.0
pillow        10.3.0
pip           23.0.1
protobuf      5.27.0
setuptools    65.5.0
six           1.16.0
sympy         1.12
tqdm          4.66.4
wxPython      4.2.1
wxtools       0.2.6.1

❯ ./bin/python app.py
/Users/kebe/projects/klipper-led_effect
Traceback (most recent call last):
  File "/Users/kebe/projects/klipper-led_effect/simulator/app.py", line 5, in <module>
    frame = simulator.Simulator(None)
  File "/Users/kebe/projects/klipper-led_effect/simulator/simulator/simulator.py", line 16, in __init__
    self.init_printer()
  File "/Users/kebe/projects/klipper-led_effect/simulator/simulator/simulator.py", line 39, in init_printer
    self.printer = mockPrinter(self.config)
  File "/Users/kebe/projects/klipper-led_effect/simulator/simulator/klippermock.py", line 19, in __init__
    self.led_effect=ledEffect(config)
  File "/Users/kebe/projects/klipper-led_effect/src/led_effect.py", line 353, in __init__
    self.layerTempl   = self.gcode_macro.load_template(config, 'layers')
AttributeError: 'NoneType' object has no attribute 'load_template'

```

julianschill, 2024-05-26:
Try the packaged version from the releases page: https://github.com/julianschill/klipper-led_effect/releases/new

julianschill, 2024-05-26:
Should be fixed now.

closed 2024-05-26 by julianschill

imkebe, 2024-05-26:
@julianschill  Where is the release?

julianschill, 2024-05-26:
Go here and scroll down to "Assets". https://github.com/julianschill/klipper-led_effect/releases

I didn't do a new release for the hotfix, though.