julianschill/klipper-led_effect #205: led_effects running on strips connected to raspberry PI

vakeiros, 2024-06-08:
hello, I'm trying to run led_effects on my printer but instead of connecting the LEDs to my SKR Pico (low processing power) I want to connect the strips to my raspberry Pi so I can take the processing load out of my SKR pico.
I've installed klipper on the PI (linux_process or host or whatever) because I also need it for my ADXL345.
I tired several GPIO on the pi, currently I'm using GPIO18 with no success.

I have these settings in printer.cfg:

[mcu rpi]
serial: /tmp/klipper_host_mcu

and on led_effect file:

[neopixel headlights]
pin: rpi:gpio20
chain_count: 2
color_order: GRBW
initial_RED: 0.7
initial_GREEN: 0.01
initial_BLUE: 0.9
initial_WHITE: 0.0

I'm not sure what I'm doing wrong.... or maybe I can't do it like this... appreciate some help

yzeroy, 2024-06-08:
Hi, klipper in current state/version does not support neopixel signal pins directly connected to RPI (HOST) GPIO pins as stated in here [](https://www.klipper3d.org/Config_Reference.html?h=temper#neopixel)

julianschill, 2024-06-08:
As @yzeroy already stated, this is a limitation from Klipper and not something I can fix in led_effects. Therefore I will close this issue.

closed 2024-06-08 by julianschill