julianschill/klipper-led_effect #206: cant figure out file structure

slightlysketch, 2024-06-09:
so i used kiauh to install klipper on 3 machines.
ive installed a ercf/happy hare on one so i understand how to set the install script up but i cant seem to make it work,
[image]
i cant seem to figure out the correct install paths for the following script.

./install-led_effect.sh -u [-k <klipper path>] [-s <klipper service name>] [-c <configuration path>]

can anyone help me out on this, im pulling my hair out and already put my fist thru one of my extra monitors.

slightlysketch, 2024-06-09:
also no matter what i do the -c never takes and always tries to install at "/home/aaron/printer_data/config"

and i get the following message.
"/home/aaron/printer_data/config" does not exist. 
Falling back to /home/aaron/klipper_config as default."

slightlysketch, 2024-06-09:
./install-led_effect.sh -k/home/aaron/klipper -s/home/aaron/sv06_data -c/home/aaron/sv06_data/config

this is what im tryig to run.

slightlysketch, 2024-06-09:
any help would be cool
id like to use the led i got but honestly will just run the original non barf one if no one can help.
always hated being poor and only running a single pi for multiple printers.
always made this stuff so much harder.

julianschill, 2024-06-09:
Try this:
```
./install-led_effect.sh -k /home/aaron/klipper -s klipper-sv06 -c /home/aaron/sv06_data/config
```

Note the spaces before the paths.

slightlysketch, 2024-06-09:
youre a god and king among men.

thank you,

julianschill, 2024-06-09:
So it worked? You might have to restart all klipper instances.