julianschill/klipper-led_effect #208: Simulator only showing line

kevinkhill, 2024-06-10:
I got the simulator up and running, but was saddened when the layout did not switch to circle.

I have a 16 pixel ring and want to experiment, but the only layout that works in the simulator for me is `line`. Circle, Triangle and Voron all do not work for me.

kevinkhill, 2024-06-10:
I think I might have found the issue! The difference between the x,y in Rectangle vs all the others, is that they were `int`

I tried adding a method like this
```python
def _push_led_coordinates(self, x, y, size, round_xy=False):
    if round_xy is True:
        self.led_coordinates += [(round(x), round(y), size)]
    else:
        self.led_coordinates += [(x, y, size)]
```
Then switched out `self.led_coordinates += [(x, y, size)]` in `_calc_coordinates()` to use that method.
With rectangle, I didn't use the 4th param to round, but enabling it in Circle worked!
```python
elif self.m_cbLayout.GetValue() == "Circle":
    self.m_spinLines.Enable(False)
    r = distance * self.led_count / (2 * math.pi)
    for i in range(self.led_count):
        a = 2 * math.pi / self.led_count * i
        x = r * math.sin(a)
        y = r * math.cos(a)
        self._push_led_coordinates(x, y, size, round_xy=True)
```
[image]

(Triangle seems to have a slight issue in the algorithm leaving gaps at certain counts)
[image]
[image]

julianschill, 2024-07-11:
I can't reproduce your problem. All shapes are working fine for me. For the triangle: You just can't form a equilateral triangle with any number of leds. That's why there are gaps.
[image]
[image]
[image]

julianschill, 2024-07-11:
Can you maybe elaborate what "not working" means for you? What behaviour are you expecting vs. what 
are you getting?

kevinkhill, 2024-07-11:
When I would try to switch to the other shapes, it would just show nothing up in the top area

julianschill, 2024-07-11:
Are you using the packaged version or are you running it with your own python? If the latter, what python version are you using?

julianschill, 2024-10-13:
Is this still an issue or can I close this?

kevinkhill, 2024-10-13:
Go ahead and close. Thanks for asking 😊

closed 2024-10-13 by julianschill