julianschill/klipper-led_effect #212: Print procress setting in LED_effect

Maxshapez, 2024-07-15:
@julianschill Hi Julianschill, I am trying to use LED_efects created by Julius. however, in my scenario I would want to explicitly set progress for progress effect rather than polling the progress value from printer state object. I tried using M73 PXX command to override the progress number but this seems have only temporary effect and the LED are reset once the print progress value is re-polled from the printer status object. Kindly advice.

julianschill, 2024-07-15:
LED effects gets the progress from the printer status. M73 sets the printer status. 
Check if you have M73 commands in your gcode file from the slicer that override your setting.

Maxshapez, 2024-07-15:
we checked that there is nothing overriding the M73 bit when we are executing it after sometime the value is resetting itself

julianschill, 2024-07-15:
Ok, but this is a Klipper issue then. LRD effects just reads the progress from the display_status object. Nothing else.

Maxshapez, 2024-07-15:
Isn't it possible to parameterized the print progress and passed it as parameter

julianschill, 2024-07-15:
It is, with M73. I'm not sure what resets it on your printer. As far as I know the print progress is ONLY set by M73. But I might be wrong.

julianschill, 2024-07-15:
At least the Klipper documentation says that the print_progress variable contains the last value from M73. And that's what LED effects uses. https://www.klipper3d.org/Status_Reference.html#display_status
So check your slicer and macros for custom gcode containing M73.

Maxshapez, 2024-07-15:
Iam not checking this during printing  i just trying it manually through mainsail when we try M73 PXX value it is working but it is resetting with in 5-10 secs 
 ( I am in Initial phase of testing LED_effect and finalizing  the color pattern in it)

julianschill, 2024-07-15:
Looking at the Klipper code here https://github.com/Klipper3d/klipper/blob/master/klippy/extras/display_status.py

Klipper resets the progress after 5 seconds when there is no print running.

closed 2024-09-29 by julianschill