julianschill/klipper-led_effect #215: Nozzle LEDs flickering if set to white + logo LED effect

bthubs, 2024-07-23:
I am currently using a Xol toolhead and wired the LED harness as per their [instructions](https://github.com/Armchair-Heavy-Industries/Xol-Toolhead/blob/main/docs/led_harness.md). It uses 1x WS2812B 5050 and 2 WS2812B 3535 (small RGB LED sequins). It works with your typical (static) stealthburner cfg. But since I am using LED effect, there is an issue such that if the logo LED has an LED effect and I set the nozzle LEDs to white, they begin flickering, But if both the nozzle and logo LEDs have an LED effect, they work just fine. I have made a few observations while troubleshooting. I've redid my harness like 5 times as I initially thought there was a loose wire touching something, but each time has been the same effect. Another thing I did notice is that if I set the framerate of the logo LED to something high, the white nozzle LEDs flicker faster, and if I set framerate lower like 2, it flickers slower. This only happens if nozzle LEDs are set to white (ie red=1 green=1 blue=1) . It does not flicker if I have a logo effect if I set the nozzle to a single color (R/G/B), but the moment I introduce another color, it flickers. I've also tried virtual_leds as well as putting a 470ohm resistor and it still doesn't work. Both the 5050 LED and 3535 LEDs are GRB color order. 

Maybe this is just a hardware issue (ie the small RGB sequins aren't compatible with LED effect?) but it doesn't make sense why nozzle + logo both with LED effect works fine, but nozzle with white and any logo effect doesn't work.  I have since changed my cfg to only have white nozzle LEDs while printing, with a static logo LED but I'm hoping a brighter mind can figure this out.

`[neopixel sb_leds]
pin: nhk:gpio7
#   The pin connected to the neopixel. This parameter must be provided.
chain_count: 3
#   The number of Neopixel chips that are "daisy chained" to the
#   provided pin. The default is 1 (which indicates only a single
#   Neopixel is connected to the pin).
color_order: GRB
#   Set the pixel order required by the LED hardware. Options are GRB,
#   RGB, GRBW, or RGBW. The default is GRB.
initial_RED: 0
initial_GREEN: 0
initial_BLUE: 0
#   Sets the initial LED color of the Neopixel. Each value should be
#   between 0.0 and 1.0. The WHITE option is only available on RGBW
#   LEDs. The default for each color is 0.#

[virtual_leds nozzle_leds]
leds: neopixel:sb_leds (2,3)

[virtual_leds logo_leds]
leds: neopixel:sb_leds (1)

##########################
# LED Effects Animations #
##########################

##################
## logo effects ##
##################

[led_effect sb_logo_busy]
autostart:              false
frame_rate:             24
leds:
    virtual_leds:logo_leds
layers:
    breathing  3 1 top (1.0, 0.0, 0.0)

[led_effect sb_logo_cleaning]
autostart:              false
frame_rate:             24
leds:
    virtual_leds:logo_leds
layers:
    breathing  3 1 top (0.0, 0.02, 0.5)

[led_effect sb_logo_calibrating_z]
autostart:              false
frame_rate:             24
leds:
    virtual_leds:logo_leds
layers:
    breathing  3 1 top (0.0, 0.0, 0.35)

[led_effect sb_logo_heating]
leds:
    virtual_leds:logo_leds
autostart:                          false
frame_rate:                         24
layers:
     breathing  3 1 top (1, 0.18, 0.0)

[led_effect sb_logo_cooling]
leds:
    virtual_leds:logo_leds
[... 264 lines cut ...]
    SET_LED_EFFECT EFFECT=sb_nozzle_homing

[gcode_macro status_cleaning]
gcode:
    STOP_LED_EFFECTS
    SET_LED_EFFECT EFFECT=sb_logo_cleaning
    SET_LED_EFFECT EFFECT=sb_nozzle_cleaning

[gcode_macro status_meshing]
gcode:
    STOP_LED_EFFECTS
    SET_LED_EFFECT EFFECT=sb_logo_meshing
    SET_LED_EFFECT EFFECT=sb_nozzle_meshing

[gcode_macro status_calibrating_z]
gcode:
    STOP_LED_EFFECTS
    SET_LED_EFFECT EFFECT=sb_logo_calibrating_z
    SET_LED_EFFECT EFFECT=sb_nozzle_calibrating_z

[gcode_macro status_printing]
gcode:
    STOP_LED_EFFECTS
    SET_LED_EFFECT EFFECT=sb_logo_printing
    SET_LED_EFFECT EFFECT=sb_nozzle_printing`

julianschill, 2024-07-24:
I have to do some tests to reproduce that and see what's causing it. 
I don't use the normal SET_LED anymore on my printer and just do static effects when I want a single color. Maybe this works for you as well.

bthubs, 2024-07-24:
> I have to do some tests to reproduce that and see what's causing it. I don't use the normal SET_LED anymore on my printer and just do static effects when I want a single color. Maybe this works for you as well.

Do you mind sharing an example of what you mean?

julianschill, 2024-07-24:
Check you effect "set_nozzle_leds". That switches the nozzle leds on using an effect. So use that instead 9f just turning them on in Mainsail.

bthubs, 2024-07-24:
I use fluidd and I dont change the colors from the interface, I usually use a macro. Is there anything wrong with my macros?

Either way, I'm suspecting a problem with the nitehawk 36 since the problem gets worse when heating up or extruding. I'll connect the harness directly to the motherboard and get back to you.

julianschill, 2024-07-24:
What macro are you using specifically? But there is nothing wrong and it should basically work. Might be a problem with updating the neopixels. Can you please post a klippy.log?

bthubs, 2024-07-24:
[klippy.log](https://github.com/user-attachments/files/16368046/klippy.log)

Sorry not sure where to look in this log, its so long... I wasn't getting any errors however. I plugged it into the motherboard (manta m8p). It's showing the same behavior, logo (LED effect - breathing) + nozzle LEDs (static white) = flickering. So it's not the Nitehawk 36. Perhaps it is the 3535 sequins then...?

DrPacemaker, 2024-12-15:
I just made the same experience following the example.cfg provided by this repo (3 LEDs). I figured out that the combination of gradient effect on logo + static full white on nozzle leds leads to this flickering. 

When changing this [code line](https://github.com/julianschill/klipper-led_effect/blob/56d1d48296f8242496d47d4b4c84c558a7bab2a5/examples/Voron_Stealthburner/stealthburner_led_effects_3_leds.cfg#L155) to e.g. `static 0 0 top (0.3, 0.1, 0.0)` the issue disappears. 

My assumption is that the led chain is struggling to change the color of LED 1 on a high frequency but keeping LED 2 & 3 on full power without interrupts, or that there is maybe a bug in the code which I don't see. 

Workaround: Avoid gradient & static effects in one led chain (at least for SB leds). 
ToDo's: Check code for this combination and adapt example.cfg to avoid guiding new joinees to this issue ;)

JanB97, 2025-05-05:
Where you able to have a look at this? 
I have the same or at least a very similar issue on two printers.

At first I thougt it is the powersupply of my V0, but I installed a rainbow barf led on my Trident yesterday and it has the same issue.

While running the "status_printing" effect and extruding, the logo and nozzle leds start flickering as long as the extruder is extruding. 
This is also happening while printing on both printers, but I am not sure if its just the extrude that is causing it or also the movement.

I have a led strip on both printers, on one of them it is configured as "output_pin" and on the other one as "led".
On both the neopixel on the display is configured as "neopixel".

Both printers are running toolhead boards, one a mellow sb2040 with CAN and the other a ldo notehawk 36 with usb.