julianschill/klipper-led_effect #222: support for effect led parameters

onsimon, 2024-09-29:
Would love to be able to configure a generic/reusable effect that can be applied to different leds using parameters or similar. Reading through the docs I don't think this is possible currently?

For example something like the below or a predefined `LEDS` parameter on `SET_LED_EFFECT` that accepts comma-separated values.

```
[led_effect panel_idle]
autostart:              true
frame_rate:             24
leds:
    neopixel:{params.leds}
layers:
    breathing  10 1 top (.5,.5,1)
```

julianschill, 2024-09-29:
Correct. This is currently only possible for layer parameters, not for setting LEDs. Implementing this is related to #24

onsimon, 2024-09-30:
@julianschill maybe another more simple solution could be wildcard or list `leds` setting (like `leds: *` or `leds: neopixel:a|neopixel:b`) for `leds` that creates instances of the effect for all or the specified leds with the led name as suffix?

onsimon, 2024-10-28:
Two more ideas/alternative workarounds that works with the limitation of one klipper object per config object @julianschill :

1. Separate effect and led_effect config objects (i.e. universal effect that can be applied to multiple leds separately, each led_effect has reference to one or more effects and one or more leds)
2. A copy led effect config object that references an existing led effect and just replaces/overwrites the led setting