julianschill/klipper-led_effect #223: Recent Klipper (v0.12.0-315-g8f361a15) Update broke ur led_effect.py

Zarrsito, 2024-09-30:
Klipper  v0.12.0-315-g8f361a15 creates the following error after the Update : Internal error during ready callback: 'LEDHelper' object has no attribute 'get_led_count', seems they changed something on the official led.py, needed to disable your plug-in atm

julianschill, 2024-09-30:
Ok, thanks for the heads-up. Will look into it ASAP.

Zarrsito, 2024-09-30:
[image]
thats why i think they changed something , i just saw neopixel.py got changes too maybe thats involved too

julianschill, 2024-09-30:
Yes, some functions the module used have been removed. Needs some fixing.

KD-4-Lyfe, 2024-10-01:
I got the same issue as well.

My Temporary fix was uncomment my neopixel.cfg which holds all of my macros and led settings.

julianschill, 2024-10-01:
Should be fixed now. Just update LED effects.

closed 2024-10-01 by julianschill

simplified-life, 2024-10-02:
Thanks for the quick response! I didn't have it in my update script. For those who forgot what they did whenever you installed it... you can just remove the old directory and install it again. https://github.com/julianschill/klipper-led_effect

aamott, 2024-10-02:
Thanks for the quick fix! Looks like there's still remnant of get_led_count somewhere because I can create a neopixel, but I get the error again when I uncomment this config section:

```cfg
# Turn on the led and start it breathing on idle
[led_effect panel_idle]
autostart:              true
Frame_rate:             24
heater:                 heater_bed
endstops: x, y, z, probe
leds:
    neopixel:board_neopixel
layers:
    breathing  10 1 average (.25,.25,0.5)
    heater  40 1 average    (1,0,0),(0,1,0)
    homing 3 0 top      (1,0,0),(0,1,0),(0,0,1)
```

julianschill, 2024-10-02:
@aamott That's weird. Maybe a different error? Please provide a klippy.log.

akashrpatel, 2024-10-03:
I'm getting this error after updating as well. Here is my klippy log
[klippy (2).log](https://github.com/user-attachments/files/17237959/klippy.2.log)

julianschill, 2024-10-03:
@akashrpatel you still seem to have the old version of LED effect. Make sure you update to v0.0.13 and then reboot your printer to make sure it uses it.

akashrpatel, 2024-10-03:
Thank you, I updated and it works fine now.

sudo-su-root, 2024-10-03:
Thanks for fixing, for anyone else still struggling who happens to be using klipper-virtual_leds... https://github.com/MapleLeafMakers/klipper-virtual_leds/issues/4

aamott, 2024-10-05:
Sorry for the delay, but here's my klippy.log. The specific config that breaks it again:
```cfg
# Turn on the led and start it breathing on idle
[led_effect panel_idle]
autostart:              true
Frame_rate:             24
heater:                 heater_bed
endstops: x, y, z, probe
leds:
    neopixel:board_neopixel
layers:
    breathing  10 1 average (.25,.25,0.5)
    heater  40 1 average    (1,0,0),(0,1,0)
    homing 3 0 top      (1,0,0),(0,1,0),(0,0,1)
```
[klippy(1).log](https://github.com/user-attachments/files/17267331/klippy.1.log)

Works without that config, but I started a master's and haven't had time to experiment. Possible something didn't update right?

julianschill, 2024-10-05:
@aamott It didn't update correctly. Consider reinstalling.

aamott, 2024-10-06:
That was totally it. Deleted the directory and reinstalled. When I checked. Thanks!

djepsylon, 2024-10-21:
same error, update don't worked and on v0.0.13 still not work.
only delete rep and re install solve problem, thanks for this super plugin