julianschill/klipper-led_effect #224: Cannot start Klipper

jankorous, 2024-09-30:
Unfortunately I had to reinstall Klipper today. After updating everything and installing all the components I used before installing, I ran into problems with LED Effects:

```
Unhandled exception during ready callback
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/klippy.py", line 164, in _connect
    cb()
  File "/home/biqu/klipper/klippy/extras/led_effect.py", line 397, in _handle_ready
    for i in range(ledChain.led_helper.get_led_count()):
AttributeError: 'LEDHelper' object has no attribute 'get_led_count'
Transition to shutdown state: Internal error during ready callback: 'LEDHelper' object has no attribute 'get_led_count'
```

led_effect in v0.0.12-0-g03a46eab version, 
klipper v0.12.0-315-g8f361a15,
mainsail v2.12.0.

ken4157, 2024-09-30:
https://github.com/julianschill/klipper-led_effect/issues/223 

the led_effect.py needs to be edited, it is best to disable led_effect in your config file until led_effect.py has been edited

czechbol, 2024-09-30:
FYI, if you are familiar with the command line and git, you could 

```git reset --hard ef75346861fdf94d952a4e887fb843d215b00631```

in the klipper folder to reset klipper to the state just before the breaking change and don't update it until this module gets patched.

Just be careful. As a rule of thumb, if you don't understand what I wrote, just # the neopixel config as @Zarrsito mentioned. It's easier to fix than a broken klipper installation.

rmilyard, 2024-09-30:
I am having same issue but not sure how to fix this.

julianschill, 2024-09-30:
Either roll back the Klipper update or comment out all led_effect instances in your config.

rmilyard, 2024-09-30:
> Either roll back the Klipper update or comment out all led_effect instances in your config.

I commented it out for the  moment.

ken4157, 2024-10-01:
https://www.facebook.com/100007992853001/videos/549489760803038/

roll back your update until it is resolved

closed 2024-10-01 by julianschill

thetic, 2024-10-02:
I'm still experiencing failures:

```
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 130, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 123, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 112, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/virtual_leds.py", line 103, in load_config_prefix
    return PrinterVirtualLeds(config)
  File "/home/pi/klipper/klippy/extras/virtual_leds.py", line 41, in __init__
    pled = printer.load_object(config, "led")
  File "/home/pi/klipper/klippy/klippy.py", line 111, in load_object
    raise self.config_error("Unable to load module '%s'" % (section,))
configparser.Error: Unable to load module 'led'
webhooks client 547455185056: New connection
webhooks client 547455185056: Client info {'program': 'Moonraker', 'version': 'v0.9.3-0-g71f9e67'}
```

julianschill, 2024-10-02:
@thetic Your error is not related to LED effects, but to "Virtual LEDs".

thetic, 2024-10-02:
My mistake. Thank you