julianschill/klipper-led_effect #228: MCU 'mcu' shutdown: Timer too close

BenJule, 2024-11-09:
I get occasionally an error with klipper_led_effect:
```
MCU 'mcu' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

[klippy (25).log](https://github.com/user-attachments/files/17686414/klippy.25.log)

julianschill, 2024-11-09:
Try deactivating all effects and check if the problem goes away. Also check the CPU load with top during a print with and without effects.

BenJule, 2024-11-09:
Done. i have deaktivate all led effects. I get no Timer too close error.

matthoskins1980, 2024-11-14:
I'm also experiencing this issue.  Deactivating led effects makes the problem go away.

BenJule, 2024-11-14:
The error usually occurs during homing or QGL. However, I have also had cases where it appeared during long prints lasting over 8 hours.

julianschill, 2024-11-16:
Timer too close errors usually indicate, that something is overloaded. It can be the host (e.g. Raspberry Pi), the communication channel to the MCU or maybe even the MCU itself. 
LED_effects need to send a lot of data to the MCU to update the LEDs. When the MCU has to handle a lot of motor steps at the same time, this error might occur. There is not much I can do about it, but to reduce the load on the system there are a few things you could do:

- Reduce the framerate of the effects.
- Only use static effects while moving
- Offload the led effects to a separate MCU by connecting the LEDs to a separate board (e.g. a Raspberry Pi Pico)
- Reduce the microsteps of the motors (I wouldn't do that just to make the LEDs work...)

Also check this guide to see if something else is causing the errors or contributes to a high system load: https://canbus.esoterical.online/troubleshooting/timer_too_close.html

jeji, 2024-12-06:
Spent two months on the same issue, and checked all the information i can found in google. 
After deactive the led efforts, issue not meet again.

NameOfTheDragon, 2024-12-12:
> Timer too close errors usually indicate, that something is overloaded. It can be the host (e.g. Raspberry Pi), the communication channel to the MCU or maybe even the MCU itself. LED_effects need to [...]

This might be related to #128 . I don't believe this is as straightforward as simple saturation since it never seems to happen during normal printing. Klipper seems much more sensitive to this (as others have pointed out) during homing, QGL or bed probing. I think the combination of LED updates and homing/probing is fatal. I wonder if it is possible to suspend LED updates while homing/probing is in progress?

julianschill, 2024-12-12:
> I wonder if it is possible to suspend LED updates while homing/probing is in progress?

Sure. Just deactivate all effects or just activate static effects. When the color of the leds are not changing, no updates are sent.

julianschill, 2024-12-12:
I had also TTC errors while doing QGL, but in my case it was camera streamer (crowsnest), that uses more and more CPU the longer it runs. Restarting that solved it for me. Check with `top` on the console if you also have a process using a lot of CPU.

smweil, 2025-01-07:
Can confirm this happens with effects on during QGL and became significantly worse with newer Klipper updates...

igeighty, 2025-01-11:
i run a manta m8 with a CB2 would a beefier board help here CM4 maybe ? or would we have to offload the led effects to another raspberry?
i cant run klipper-effects without a timer to close error :(

Didier-T, 2025-01-15:
Bonjour,
Je rencontre aussi le souci de minuteur trop proche, uniquement lors de l'utilisation de led_effect.
J'ai l'impression que le souci est moins fréquent quand je baisse le taux de rafraichissement.
Pour information je suis aussi avec une m8p et un cb2, ma tête d'outils est branché en canbus.

J'adore votre travaille et j'espère que vous réussirez a trouver une solution

KD-4-Lyfe, 2025-01-31:
> i run a manta m8 with a CB2 would a beefier board help here CM4 maybe ? or would we have to offload the led effects to another raspberry? i cant run klipper-effects without a timer to close error :(

Mine is a manta m8p CB1

I have the same issues, I did try use the CB2 but i had alot of compatability issues with the CB2 and newest version of klipper. im about to send mine back and get a CM4 or CM5 if theres any in my area yet.

My other option was to use a pi pico as i have a few.

NameOfTheDragon, 2025-02-01:
FYI upgrading to a Raspberry Pi 4 has solved all my issues. Some people have suggested that this problem is worse on Raspberry Pi 3 with 64-bit Raspian OS. That was indeed what I was running so I upgraded to Raspberry Pi 4 and it is a night-and-day difference. I was motivated to upgrade when I installed an ERCF MMU and the Happy Hare firmware caused a lot of shutdowns. So anyone having this issue, get yourself a Raspberry Pi 4 (now that the 5 is out, the price should drop!).