julianschill/klipper-led_effect #233: Brightness for Temp isnt changing.

DenDovne, 2024-12-08:
Temperature
Effect Rate:  20  Cold Temperature
Cutoff:       80  Hot Temperature
Palette:          Color values to blend from "Cold" to "Hot"
The temperature of the configured heater determines the color in a gradient over the palette. When only one color is defined in the palette, the brightness of that color is defined by the temperature.

[led_effect bed_heater_leds]
leds:
    neopixel:bed_lights (1-20)
autostart: false
frame_rate:24
heater: heater_bed
layers:
    heater 25.00 0.00 add (0,0,0),(1,0,0),(1.00,0.00,0.00)

its staying at one brightness until it hits the final temp, then changes to final temp color

julianschill, 2024-12-09:
This should actually work and start getting brighter at 28° until the bed reaches 70°. But you maybe want to use the heater effect instead, as it uses the current target temperature instead of a fixed one.

There might be an issue updating the leds. Can you try making an animated effect (like gradient or chase) and see if it keeps moving while the bed is heating?

DenDovne, 2024-12-13:
im going to try it with chase and see if it does. when i did the heater one, or the temp one it does go from blue (cold) to red (hot) and final red, same as the end temp, it does start blue and fades to red, goes from blue to purple, then to red, and it still goes from like a deep purple to the solid red in one big jump, the last fades i feel like get skipped, not gradual like it is heating up. ideally i want to just use red and have it start at like 20-25C and slowly fade to the final red full brightness.