julianschill/klipper-led_effect #234: Multipule Tool heads

Cagle3545, 2024-12-11:
I am trying to figure out how to configure for multipule tool heads. Each toolhead has 3 LEDs and their own LED Pin. I am able to get one tool head to take the effects, but I am unable to figure out how to get the effect to go to multipule toolheads. I would like to be able to have the current selected toolhead to take the current effect and the idle toolheads to take on other effects depending on what that toolhead is doing, ie heating up, sitting idle...

julianschill, 2024-12-11:
Currently an effect is bound to dedicated leds. So you have to define the effects for each toolhead individually. There are plans to make it possible to change the leds on runtime, but that is not implemented yet. See #24 and #231

rmerli, 2026-03-14:
Hello, I ran into a very similar limitation with my StealthChanger setup.

My goal was to have the logo LED on each toolhead reflect the temperature of its hotend using a gradient:
- when cold, the LED should be static
- while heating or cooling, it should breathe
- once the target temperature is reached, it should become static again

I was able to get something close with the current plugin, but the main issue for me was that the breathing effect was not synchronized across toolheads, since I had to define one effect per toolhead.

Because of that, I hacked together a proof of concept on my fork. It is definitely rough and not something I would consider production-ready yet, but I wanted to share it to get feedback on whether this direction makes sense at all.

With my changes, this is the config needed to achieve the behavior described above:

```ini
[led_effect multi_tool_breathing]
autostart: true
frame_rate: 24
heater_map:
    extruder=neopixel:T0 (1)
    extruder1=neopixel:T1 (1)
    extruder2=neopixel:T2 (1)
    extruder3=neopixel:T3 (1)
leds:
layers:
    multiheater 60 target-2 multiply (0.0,1.0,1.0),(0.0,1.0,1.0),(0.0,1.0,1.0),(0.0,0.0,1.0),(0.0,1.0,0.0),(1.0,1.0,0.0),(1,0.30,0.0),(1,0.15,0.0),(1,0.125,0.0),(1.0,0.1,0.0)
    breathing 2 0 top (1.0,1.0,1.0)

[led_effect multi_tool_ready]
autostart: true
frame_rate: 24
heater_map:
    extruder=neopixel:T0 (1)
    extruder1=neopixel:T1 (1)
    extruder2=neopixel:T2 (1)
    extruder3=neopixel:T3 (1)
leds:
layers:
    multiheater target-2 target+2 top (0.0,1.0,1.0),(0.0,1.0,1.0),(0.0,1.0,1.0),(0.0,0.0,1.0),(0.0,1.0,0.0),(1.0,1.0,0.0),(1,0.30,0.0),(1,0.15,0.0),(1,0.125,0.0),(1.0,0.1,0.0)

[led_effect multi_tool_base]
autostart: true
frame_rate: 24
heater_map:
    extruder=neopixel:T0 (1)
    extruder1=neopixel:T1 (1)
    extruder2=neopixel:T2 (1)
    extruder3=neopixel:T3 (1)
leds:
layers:
    multiheater 0 60 top (0.0,1.0,1.0),(0.0,1.0,1.0),(0.0,1.0,1.0),(0.0,0.0,1.0),(0.0,1.0,0.0),(1.0,1.0,0.0),(1,0.30,0.0),(1,0.15,0.0),(1,0.125,0.0),(1.0,0.1,0.0)
```

The basic idea is to define a mapping of `extruder -> led`.

Right now, this mapping is only used by a new `multiheater` layer.

Syntax:

```ini
multiheater {min_temp} {max_temp} {blend_mode} {palette}
```

I also made `{min_temp}` and `{max_temp}` accept a `target` variable, so temperatures can be defined relative to the heater target. In theory this supports `+`, `-`, `/`, and `*`, so simple math expressions should work.

Examples:

```text
target=200, "target+1" -> 201
target=200, "target/2" -> 100
```

## Notes / limitations

- You still need to specify the `leds:` field because there is currently validation that requires it, but it should be left empty. `heater_map` is what is actually used.
- I only tested this with the `breathing` layer. In theory it should work with other non heater related layers but I have not tested that.
- This was built specifically to solve a problem I had, so if you decide to test it keep in mind it might break your setup.

I mainly wanted to share this to see whether this kind of approach makes sense for the plugin.

Link to the branch:

https://github.com/rmerli/klipper-led_effect/tree/multi-tool-heater-effect