julianschill/klipper-led_effect #237: Heater effect animation

DVSVIDEO, 2025-01-03:
It takes 5-8 minutes to heat the bed. The heating effect turns on 22 LEDs at a time.
Very useful effect, but I find it very static.
I would like to suggest that you add an animation that would run a less brightly lit LED from the first to the last one that is already lit.
Similar to the animation of the progress bar in windows, letting you know there is a change even if there is no change in % at the moment.

julianschill, 2025-02-19:
That's what the layers are for. Take the heater effect for the color and choose an animation with white colors. Then multiply them. 

Example:
`chase 1.00 4.00 multiply (1.00,1.00,1.00),(0.94,0.94,0.94) 
heater 1.00 1.00 top (0.0,0.0,1.0),(1.0,0.0,0.0),(1.0,0.0,0.0) 
`

With heatergauge you can make an animated progress bar, that also changes color:

`heatergauge 100.00 0.00 multiply (1.00,1.00,1.00),(1.00,1.00,1.00) 
chase 1.00 4.00 multiply (1.00,1.00,1.00),(0.94,0.94,0.94) 
heater 1.00 1.00 top (0.0,0.0,1.0),(1.0,0.0,0.0),(1.0,0.0,0.0) `

[image]

DVSVIDEO, 2025-03-01:
Thank you for your reply, this is not what I had in mind. 
At the moment, I have solved it so that all the LEDs are white at the beginning (this is the default before printing), and then after turning on the bed heater, the white LEDs turn red one by one.
My idea was that the increasing number of red LEDs would not be static, but that every second or so a fainter red would run from the beginning to the point where the heating is on, where the LEDs would be red instead of white (in windows the stream indicator strip not only increases, but pulses)

julianschill, 2025-06-16:
@borodihin Set the second parameter (cutoff) to 0. It will then switch to the last color of the palette when the temperature is reached (you might want to copy your last color).