julianschill/klipper-led_effect #254: installation on Creality K1 / MIPS systems

nicochi, 2025-09-04:
See below chances to "install-led_effect.sh" for installation on Creality K1/MIPS systems inspired by Happy Hare install script. Changes at the beginning of the file to set paths and in functiuon verify_ready()

```shell
#!/bin/bash
# Force script to exit if an error occurs
set -e

# Creality K1 series printers run on MIPS, with a limited instruction set and different default klipper directories
# Checking for machine type is the easiest way so far to spot them (will be set to 1 if on MIPS):
IS_MIPS=0
if [ $(uname -m) = "mips" ]; then
    IS_MIPS=1
fi

KLIPPER_PATH="${HOME}/klipper"
KLIPPER_SERVICE_NAME=klipper
SYSTEMDDIR="/etc/systemd/system"
MOONRAKER_CONFIG_DIR="${HOME}/printer_data/config"

if [ "$IS_MIPS" -eq 1 ]; then
    KLIPPER_PATH="/usr/share/klipper"
    MOONRAKER_CONFIG_DIR="/usr/data/printer_data/config"
fi

# Fall back to old directory for configuration as default
if [ ! -d "${MOONRAKER_CONFIG_DIR}" ]; then
    echo "\"$MOONRAKER_CONFIG_DIR\" does not exist. Falling back to "${HOME}/klipper_config" as default."
    MOONRAKER_CONFIG_DIR="${HOME}/klipper_config"
fi

usage(){ echo "Usage: $0 [-k <klipper path>] [-s <klipper service name>] [-c <configuration path>] [-u]" 1>&2; exit 1; }
# Parse command line arguments
while getopts "k:s:c:uh" arg; do
    case $arg in
        k) KLIPPER_PATH=$OPTARG;;
        s) KLIPPER_SERVICE_NAME=$OPTARG;;
        c) MOONRAKER_CONFIG_DIR=$OPTARG;;
        u) UNINSTALL=1;;
        h) usage;;
    esac
done

# Find SRCDIR from the pathname of this script
SRCDIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )"/src/ && pwd )"

# Verify Klipper has been installed
check_klipper()
{
    if [ "$IS_MIPS" -ne 1 ]; then
      if [ "$(sudo systemctl list-units --full -all -t service --no-legend | grep -F "$KLIPPER_SERVICE_NAME.service")" ]; then
          echo "Klipper service found with name "$KLIPPER_SERVICE_NAME"."
      else
          echo "[ERROR] Klipper service with name "$KLIPPER_SERVICE_NAME" not found, please install Klipper first or specify name with -s."
          exit -1
      fi
    else
      # There is no systemd on MIPS, we can only check the running processes
      running_klipper_pid=$(ps -o pid,comm,args | grep [^]]/klipper/klippy/klippy.py | awk '{print $1}')
      KLIPPER_PID_FILE=/var/run/klippy.pid

      if [ $(cat $KLIPPER_PID_FILE) = $running_klipper_pid ]; then
        echo -e "${DIM}Klipper service found"
      else
        echo -e "${ERROR}Klipper service not found! Please install Klipper first"
        exit -1
      fi
    fi
  
}

check_folders()
{
    if [ ! -d "$KLIPPER_PATH/klippy/extras/" ]; then
        echo "[ERROR] Klipper installation not found in directory \"$KLIPPER_PATH\". Exiting"
        exit -1
    fi
    echo "Klipper installation found at $KLIPPER_PATH"

    if [ ! -f "${MOONRAKER_CONFIG_DIR}/moonraker.conf" ]; then
        echo "[ERROR] Moonraker configuration not found in directory \"$MOONRAKER_CONFIG_DIR\". Exiting"
        exit -1
    fi
    echo "Moonraker configuration found at $MOONRAKER_CONFIG_DIR"
}

# Link extension to Klipper
link_extension()
{
    echo -n "Linking extension to Klipper... "
    ln -sf "${SRCDIR}/led_effect.py" "${KLIPPER_PATH}/klippy/extras/led_effect.py"
    echo "[OK]"
}

# Restart moonraker
restart_moonraker()
{
    echo -n "Restarting Moonraker... "
    sudo systemctl restart moonraker
    echo "[OK]"
}

# Add updater for led_effect to moonraker.conf
add_updater()
{
    echo -e -n "Adding update manager to moonraker.conf... "

    update_section=$(grep -c '\[update_manager led_effect\]' ${MOONRAKER_CONFIG_DIR}/moonraker.conf || true)
    if [ "${update_section}" -eq 0 ]; then
        echo -e "\n" >> ${MOONRAKER_CONFIG_DIR}/moonraker.conf
        while read -r line; do
            echo -e "${line}" >> ${MOONRAKER_CONFIG_DIR}/moonraker.conf
        done < "$PWD/file_templates/moonraker_update.txt"
        echo -e "\n" >> ${MOONRAKER_CONFIG_DIR}/moonraker.conf
        echo "[OK]"
        restart_moonraker
        else
        echo -e "[update_manager led_effect] already exists in moonraker.conf [SKIPPED]"
    fi
}

restart_klipper()
{
    echo -n "Restarting Klipper... "
    sudo systemctl restart $KLIPPER_SERVICE_NAME
    echo "[OK]"
}

start_klipper()
{
    echo -n "Starting Klipper... "
    sudo systemctl start $KLIPPER_SERVICE_NAME
    echo "[OK]"
}

stop_klipper()
{
    echo -n "Stopping Klipper... "
    sudo systemctl stop $KLIPPER_SERVICE_NAME
    echo "[OK]"
}

uninstall()
{
    if [ -f "${KLIPPER_PATH}/klippy/extras/led_effect.py" ]; then
        echo -n "Uninstalling... "
        rm -f "${KLIPPER_PATH}/klippy/extras/led_effect.py"
        echo "[OK]"
        echo "You can now remove the [update_manager led_effect] section in your moonraker.conf and delete this directory. Also remove all led_effect configurations from your Klipper configuration."
    else
        echo "led_effect.py not found in \"${KLIPPER_PATH}/klippy/extras/\". Is it installed?"
        echo "[FAILED]"
    fi
}

# Helper functions
verify_ready()
{
    if [ "$IS_MIPS" -eq 0 ]; then
      if [ "$EUID" -eq 0 ]; then
          echo "[ERROR] This script must not run as root. Exiting."
          exit -1
      fi
    fi
}

# Run steps
verify_ready
check_klipper
check_folders
stop_klipper
if [ ! $UNINSTALL ]; then
    link_extension
    add_updater
else
    uninstall
fi
start_klipper
```

nicochi, 2025-09-04:
Note that path in /file_templates/moonraker_update.txt or after installating in moonraker.conf need to be changed to 

```shell
path: /usr/data/printer_data/config/klipper-led_effect
```

julianschill, 2025-09-05:
Wouldn't it be possible to just pass the directories as command line parameters? Then a documentation entry would suffice.

nicochi, 2025-09-05:
Sure parameters should work, but if you like to implement this small change will be easier than to figure out the correct paths... and for some reason Klipper on Creality is triky... especially firmware K1 2.3.5.33 as comes without bash. I installed bash but did not manage to pass parameters sucesfully and modifying the script was easier for me.